MicrosoftDocs/windows-driver-docs #990: I can't download the wdk

xialiangfeng, 2018-11-02:
[Enter feedback here]

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 3f5d6d55-cb35-6b9e-f8ca-e75f70826dc3
* Version Independent ID: 1d913516-b1e0-e872-c337-e30ed5dffce8
* Content: [Previous WDK versions and other downloads - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/other-wdk-downloads)
* Content Source: [windows-driver-docs-pr/other-wdk-downloads.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/other-wdk-downloads.md)
* Product: **windows-hardware**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **windowsdriverdev**

xialiangfeng, 2018-11-06:
I find when I using the wdksetup.exe for win10 1803, which is downloaded from the url:"https://docs.microsoft.com/en-us/windows-hardware/drivers/download-the-wdk", it seems trying to download "windows driver kit - windows 10.0.10586.0". But  the version of win10 1803 (which I get it using vs2015) is 10.0.17134. Would this be correct?

xialiangfeng, 2018-11-06:
While the winsdksetup.exe for win10 1803 works well, I install the SDK - windows 10.0.17134.12 successfully.

tedhudek, 2018-11-14:
Closing this as duplicate of #1007

closed 2018-11-14 by tedhudek