MicrosoftDocs/windows-driver-docs #995: great article

RAJU2529, 2018-11-03:
thanks for article published by authors.
But I could not able to understand because  , I am not a programmer , to read this article, first I need to learn programming languages .

This article is very helpful for designing of gpu.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: bbeaad59-c4c0-b98d-da80-cfcda9da2ce5
* Version Independent ID: 1e213c93-4bb7-dad0-49c9-c996c73bc04a
* Content: [GpuMmu model - Windows drivers](https://docs.microsoft.com/en-in/windows-hardware/drivers/display/gpummu-model)
* Content Source: [windows-driver-docs-pr/display/gpummu-model.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/display/gpummu-model.md)
* Product: **windows-hardware**
* GitHub Login: @nabazan-microsoft
* Microsoft Alias: **windowsdriverdev**

mimisasouvanh, 2018-11-05:
@RAJU2529 
Hi Varadharajan, Thank you for you comment! Yes, that is correct, in order to develop on GPUs, you must learn driver development. You can start by learning the programming language C++, and then learn how to write drivers.
Best,
Mimi

closed 2018-11-05 by mimisasouvanh

RAJU2529, 2019-01-19:
Thanks  Mimi madam