MicrosoftDocs/windows-driver-docs #1007: when download the wdk for win10 version 1803, the setup log gets:"invalid payload".

xialiangfeng, 2018-11-05:


EliotSeattle, 2018-11-05:
Linked to Issue #990

tedhudek, 2018-11-05:
Hi @xialiangfeng sorry you are having this issue. Please try cleaning out the temp dir, remove the setup binary, and start over with a new installation.

tedhudek, 2018-11-14:
Hi @xialiangfeng sorry for the delay.  When you try to install the 1803 WDK, are you using the link from https://docs.microsoft.com/windows-hardware/drivers/other-wdk-downloads?  Specifically, the link is https://go.microsoft.com/fwlink/?linkid=873060.  I just installed that 1803 WDK with no problems.

Have you tried asking your question on one of the forums?  That is a better place to ask questions specific to the product.  Those links are:

* https://social.msdn.microsoft.com/Forums/windowshardware/en-US/home?forum=wdk
* http://www.osronline.com/cf.cfm?PageURL=showlists.CFM?list=NTDEV

Hope this helps!

closed 2018-12-14 by tedhudek