MicrosoftDocs/windows-driver-docs #1008: Such a quick way to get the tools without the entire SDK.....

bigboydrs, 2018-11-05:
http://download.microsoft.com/download/A/6/A/A6AC035D-DA3F-4F0C-ADA4-37C8E5D34E3D/setup/WinSDKDebuggingTools/dbg_x86.msi

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 162fe2fe-9216-f4c5-9f91-3df8776bb99e
* Version Independent ID: 013dbf6b-0790-9c11-24f4-dbf24d64401e
* Content: [Download Debugging Tools for Windows - WinDbg - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/debugger-download-tools)
* Content Source: [windows-driver-docs-pr/debugger/debugger-download-tools.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/debugger-download-tools.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

aluhrs13, 2018-11-05:
That links to a very old version, 6.12.2.633 from 2010. It shouldn't be used for debugging anything newer than Windows Vista.

DOMARS, 2018-11-07:
@bigboydrs - The supported download locations for the current debuggers are provided in the docs. 
Thanks for the interest in our docs and the input.

closed 2018-11-07 by DOMARS