MicrosoftDocs/windows-driver-docs #1009: API documentation links are broken.

apozsgay, 2018-11-06:
I cannot access API documentation for WinUSB isochronous. For example:
https://msdn.microsoft.com/library/windows/hardware/dn265566

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 8726446f-47fc-f7d5-790b-09790473b3f0
* Version Independent ID: 9806b8f3-0314-f89a-973f-0a9c12309e8d
* Content: [Send USB isochronous transfers from a WinUSB desktop app - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/getting-set-up-to-use-windows-devices-usb)
* Content Source: [windows-driver-docs-pr/usbcon/getting-set-up-to-use-windows-devices-usb.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/usbcon/getting-set-up-to-use-windows-devices-usb.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

tedhudek, 2018-12-13:
Thanks for the report @apozsgay but I don't repro the problem. The page in question has links to https://msdn.microsoft.com/library/windows/hardware/dn265566, which redirects to https://docs.microsoft.com/en-us/windows/desktop/api/winusb/nf-winusb-winusb_registerisochbuffer, which is the expected destination.  Note that we moved all these docs from MSDN to Docs.microsoft.com recently, so it does appear on a different site.  But it's the same content.

closed 2018-12-13 by tedhudek

apozsgay, 2018-12-13:
@tedhudek yes it appears that this was fixed sometime during the 5 weeks since I reported it.

tedhudek, 2018-12-13:
That's what I figured :)  Sorry for the delay!

has-syed, 2021-10-18:
Sample code is slightly wrong. 
The function WinUsb_WriteIsochPipeAsap() takes 5 parameters. 
There is no documentation on what one should do if WinUsb_WriteIsochPipeAsap() fails with ERROR_IO_PENDING.