MicrosoftDocs/windows-driver-docs #1010: EV Code Sign Certificates provider selection

realnickel, 2018-11-06:
Please, clarify current status on EV Code Sign Certificates selection guide for UEFI binaries signing.
In a below document :
https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/lsa-and-uefi-file-signing
there are links to 
1. Code signing certificate selection guide
https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/get-a-code-signing-certificate
2. UEFI CA Signing policy updates
https://blogs.msdn.microsoft.com/windows_hardware_certification/2013/12/03/microsoft-uefi-ca-signing-policy-updates/

The former document (issued on 
04/20/2017) includes the following approved EV Code Signing Certificates providers:

> Extended validation code signing certificates (required for UEFI, kernel-mode drivers, and LSA certifications)
> 
>     Buy a Symantec EV code signing certificate
> 
>     Buy a Certum EV code signing cert
> 
>     Buy an Entrust EV code signing cert
> 
>     Buy a GlobalSign EV code signing certificate
> 
>     Buy a Comodo EV code signing certificate
> 
>     Buy a DigiCert EV code signing certificate
> 
> 

while the latter (an older one though, issued on December 3, 2013) mentions only two:

> a. Starting March 15, 2014, SysDev will begin accepting EV certificates issued by either DigiCert or VeriSign for new submissions and renewals.

Is it Okay to use, for example, Comodo EV Code Sign Certificate to sign UEFI binary submisions as of autumn, 2018?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: b119e13e-5e0a-a598-6799-5bcd1d45cb2f
* Version Independent ID: 7bd929b5-490a-1724-152a-5172f480cf80
* Content: [LSA and UEFI file signing - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/lsa-and-uefi-file-signing)
* Content Source: [windows-driver-docs-pr/dashboard/lsa-and-uefi-file-signing.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/lsa-and-uefi-file-signing.md)
* Product: **windows-hardware**
* GitHub Login: @aahill
* Microsoft Alias: **eliotgra**

OzBob, 2019-03-06:
The 'Comodo' link has been tampered with!

The ["Microsoft link"](https://go.microsoft.com/fwlink/?linkid=863208) currently points to InstantSSL "https://www.instantssl.com/code-signing-certificates"

instead of the new owners of Comodo: https://sectigo.com/products/signing-certificates/code-signing

I would create a pull request myself and correct the Docs, but do not know how to create a new Microsoft shorturl.

EliotSeattle, 2019-03-06:
@SillyKeith, can you please look at this and Issue #1384

SillyKeith, 2019-05-30:
The link was fixed back in March - in OzBob's comment, the "Microsoft Link" takes me to Sectigo.

SillyKeith, 2019-05-30:
Yes, it's ok to use Comodo (now Sectigo) EV code signing for UEFI submissions.

tedhudek, 2019-05-30:
Thanks @SillyKeith!

closed 2019-05-30 by tedhudek

CodeSigningCertificate, 2020-05-19:
Here are the common documents and verification steps Certificate Authority follows to validate the EV Code Signing Certificate.

- Enrollment Form
- Organization Authentication
- Operational Existence
- Physical Address
- Telephone Verification
- Employment Verification
- Final Verification Call

Source: https://codesigningstore.com/documents-required-for-ev-code-signing-certificate-validation

jankipmehta, 2023-03-17:
Getting through the Extended Validation process is complex, as a minor mistake can detain you from the procedure. The primary documents you require to pass Extended Validation are as follows:

- Company Registration (Report from Dun & Bradstreet, Government License, Bank Statement, Professional Opinion Letter)
- Physical Address Proof (Utility Bill, Lease Agreement, Rent Agreement, etc.)
- Telephone Number to contact for verification (A working landline or mobile number)
- Operational Existence Document (Tax statements, Bank Statement of Demand Deposit account, etc.)

Read more about an EV code signing certificate - https://sslwiki.org/what-is-extended-validation-ssl-certificate/