MicrosoftDocs/windows-driver-docs #1013: The "Trace" marco that defintion by WDF Template cause "MYDRIVER_ALL_INFO" undefined symbol.

GreatFireWall, 2018-11-06:
place_holder for images.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: f5d3d49a-28cd-231f-d58e-743cd6bc02f5
* Version Independent ID: 7adf2d54-13be-6458-77d6-280de3464cff
* Content: [Adding WPP Software Tracing to a Windows Driver - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/adding-wpp-software-tracing-to-a-windows-driver#step-2-choose-which-trace-message-functions-you-intend-to-use-and-define-the-wpp-macros-for-those-functions)
* Content Source: [windows-driver-docs-pr/devtest/adding-wpp-software-tracing-to-a-windows-driver.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/adding-wpp-software-tracing-to-a-windows-driver.md)
* Product: **windows-hardware**
* GitHub Login: @nabazan-microsoft
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2019-04-03:
Hi @GreatFireWall 
Thank you for taking the time to share your product and documentation feedback with us. Your input is valued because it helps us create the right documentation for our customers. 
Due to the volume of issues in our queue, our policy is to close issues that have been inactive for the last 90 days. Unfortunately, we have not been able to allocate a writer to follow up on this during this time. We hope to continue hearing from you. 
Thank you.

closed 2019-04-03 by EliotSeattle

GreatFireWall, 2019-04-03:
So, settle a matter by leaving it unsettled ?

EliotSeattle, 2019-04-03:
I wish we had the staff to track down every question. Since we haven't been able to get to this particular Issue in a long time, I think it's better to be honest than to have you think someone is researching this at this time. 
I am truly sorry. I really wish we could accomodate all questions.
Eliot