MicrosoftDocs/windows-driver-docs #1016: Proofreading needed

emayevski, 2018-11-07:
It is hard to understand contents when the text is not proof-read. E.g. "Windows synchronous offload reads and writes operation." or the duplicate paragraph in "ODX Operation with One Server" section, or the phrase in the same section (2nd copy), which references some illustration (is there expected to be a picture there?). 

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 626d5b1d-401b-d9bd-1f74-0ed536bdcb4b
* Version Independent ID: 15e7bdfc-7220-9ee6-ceb5-335da59545be
* Content: [Offloaded Data Transfer - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/storage/offloaded-data-transfer)
* Content Source: [windows-driver-docs-pr/storage/offloaded-data-transfer.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/storage/offloaded-data-transfer.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

winter3angle, 2018-12-13:
I upvote this. It was disappointing to see pearls like mentioned above here and there in my case in HLK/HCK docs.

rodwiddowson, 2018-12-13:
"test in production"  With us as the testers?

lorihollasch, 2019-10-07:
@emayevski, thank you for reporting the issues with this page - it did indeed need a good number of modifications. Apologies for the slow response to this, but it's now done!

closed 2019-10-07 by lorihollasch