MicrosoftDocs/windows-driver-docs #1020: No references

rob18767, 2018-11-08:
No references under "common properties" after right clicking on package..... Honestly. It seems nothing Microsoft writes is of any use.  

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 0706bf23-aa1a-e71f-0e20-c98639d2e0f6
* Version Independent ID: ea71830e-18f6-51d9-b13d-d28079f7b411
* Content: [Creating a Driver Package - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/develop/creating-a-driver-package)
* Content Source: [windows-driver-docs-pr/develop/creating-a-driver-package.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/develop/creating-a-driver-package.md)
* Product: **windows-hardware**
* GitHub Login: @tedhudek
* Microsoft Alias: **windowsdriverdev**

tedhudek, 2018-12-14:
Driver packages are no longer created by default.  Thanks for the tip.

tedhudek, 2018-12-17:
Hi @rob18767 I've made some updates to the doc page to accommodate UI changes since that page was published.  Thanks again for bringing this to our attention.

closed 2018-12-17 by tedhudek