MicrosoftDocs/windows-driver-docs #1021: How to  enable code integrity for all the drivers

RAJU2529, 2018-11-08:
Great article 

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: c62e2f4e-197b-1d5c-485a-8c36b8476ce1
* Version Independent ID: 9f557051-4599-3c35-5ca9-edfbaaf1fa7a
* Content: [Code Integrity Diagnostic System Log Events - Windows drivers](https://docs.microsoft.com/en-in/windows-hardware/drivers/install/code-integrity-diagnostic-system-log-events)
* Content Source: [windows-driver-docs-pr/install/code-integrity-diagnostic-system-log-events.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/code-integrity-diagnostic-system-log-events.md)
* Product: **windows-hardware**
* GitHub Login: @tedhudek
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-11-08:
Thanks, @RAJU2529! It's always nice to get feedback like this. @tedhudek is a heck of a writer. ; )

sarman1998, 2018-12-03:
The info for enabling is in the "Enabling the System Event Audit Log" article.  
I suggest you update this article to make that clear, or link directly to https://docs.microsoft.com/en-us/windows-hardware/drivers/install/enabling-the-system-event-audit-log#how-to-enable-verbose-logging-of-code-integrity-diagnostic-events in the second to last paragraph.

EliotSeattle, 2018-12-05:
The link is in the article. Thanks!

closed 2018-12-05 by EliotSeattle