MicrosoftDocs/windows-driver-docs #1036: Is there a product count limitation or request interval to get all product list ?

roys93, 2018-11-12:
I cannot get product list over 1,650 even though change "pageSize" parameter in "next_link" URL.

This url is the error link to get next page.
"https://manage.devcenter.microsoft.com/v1.0/my/hardware/products?pageSize=50&continuationToken=PD94bWwgdmVyc2lvbj0iMS4wIiBlbmNvZGluZz0idXRmLTE2Ij8%2BPENvbnRpbnVhdGlvblRva2VuPjxWZXJzaW9uPjIuMDwvVmVyc2lvbj48VHlwZT5UYWJsZTwvVHlwZT48TmV4dFBhcnRpdGlvbktleT4xITQ4IWNIVmliR2x6YUdWeWN5MHdNREF3TURBd01EQXdNREF3TURBd01ESTVPVGc1T0RZdzwvTmV4dFBhcnRpdGlvbktleT48TmV4dFJvd0tleT4xITk2IWRYTmxjaTFrWld4bGRHVmtMVEF0SUNBZ0lDQWdTR0Z5WkhkaGNtVkVjbWwyWlhJdGNISnZaSFZqZEhNdE1EQXdNREF3TURBeE16WTVPVE00TkRneU9ESXpOVEE0TWctLTwvTmV4dFJvd0tleT48VGFyZ2V0TG9jYXRpb24%2BU2Vjb25kYXJ5PC9UYXJnZXRMb2NhdGlvbj48L0NvbnRpbnVhdGlvblRva2VuPg%3D%3D"

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 40f09988-3ba3-0ce2-9952-b11ddd66aff1
* Version Independent ID: dd0d280c-f6a0-4a07-7f86-42eafaf6087b
* Content: [Get all products - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/get-all-products)
* Content Source: [windows-driver-docs-pr/dashboard/get-all-products.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/get-all-products.md)
* Product: **windows-hardware**
* GitHub Login: @balapv
* Microsoft Alias: **balapv**

EliotSeattle, 2018-11-12:
@balapv, can you comment on this?

EliotSeattle, 2018-11-29:
Hi @roys93 

I asked one of the engineers on the dashboard team to look into this, and he said:

"I tried to reproduce the below issue with an account which has 2301 products in the PROD environment.

1)	With a page size of 50 results, I was able to fetch all the pages (48 pages of 50 entries).
2)	I preserved the links for all 48 pages and used them in a random order (48th page, 25th page, 37th page, 2nd page). This also worked. 

NOTE: as per this [StackOverflow entry](https://stackoverflow.com/questions/21548597/azure-table-storage-continuation-token-lifetime) there is no expiration for the Azure table storage token, and if the table storage data is updated then he may get a different set of results.

Can you please provide us the complete use case  that you have tried and your publisher id to cross check with partner specific data?"

You can reach out directly to me at eliotgra at microsoft dot com.

Thanks,
@EliotSeattle

hatasum, 2018-12-25:
I have the same question.
https://github.com/Microsoft/SDCM/issues/9
My company are over 10000 products.
I use Get all products API, only can get 8910 products.
At last times, the service shows the error : "500 - Internal Server Error".

closed 2019-03-06 by EliotSeattle