MicrosoftDocs/windows-driver-docs #1037: The link to MBIM v1.0 is no longer valid...

jefsong, 2018-11-12:
The updated link should be: https://usb.org/document-library/mobile-broadband-interface-model-v10-errata-1-and-adopters-agreement

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 1613c72c-8238-610e-8ed8-9faeff44f339
* Version Independent ID: 391c4c45-1d34-fbe1-3fea-b3eb55e279ed
* Content: [Mobile Broadband Implementation Guidelines for USB Devices - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/network/mobile-broadband-implementation-guidelines-for-usb-devices#feedback)
* Content Source: [windows-driver-docs-pr/network/mobile-broadband-implementation-guidelines-for-usb-devices.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/network/mobile-broadband-implementation-guidelines-for-usb-devices.md)
* Product: **windows-hardware**
* GitHub Login: @tedhudek
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-11-12:
Thanks, @jefsong.
Updated link. Should be live shortly.

closed 2018-11-12 by EliotSeattle

duncanmacmichael, 2018-11-12:
Thanks for taking care of that @EliotSeattle!