MicrosoftDocs/windows-driver-docs #1041: How to use the Windows online public symbols with traceview.exe

RobSpiger, 2018-11-13:
traceview.exe accepts the -i parameter in what appears to be the same way that tracepdb does.  This means you can set the environment variable _NT_SYMBOL_PATH per the docs at https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/microsoft-public-symbols and then use the -i parameter with traceview to have it search for information from the online .pdb files.  This means you don't need to have access to the symbol files locally to use traceview.exe.  Example (after setting the _NT_SYMBOL_PATH): traceview.exe -process tpm.etl -i

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: e400448d-0848-ef95-f9ed-cba41fc4b974
* Version Independent ID: 9f8e9458-bb9b-0292-0a85-97875e06e3a3
* Content: [TraceView Control Commands - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/traceview-control-commands)
* Content Source: [windows-driver-docs-pr/devtest/traceview-control-commands.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/traceview-control-commands.md)
* Product: **windows-hardware**
* GitHub Login: @nabazan-microsoft
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-11-13:
Thanks, @RobSpiger. I'm reaching out to the devs who own TraceView to verify. Will come back to this as soon as I have something.

RobSpiger, 2018-11-15:
The -i option worked on the September 2018 version of traceview.exe (visible by right clicking on the traceview.exe file and looking at the details tab).  Someone else told me it didn't work on an earlier version, so you might need to upgrade before the -i option works.

RobSpiger, 2018-11-15:
Hello-

FYI - I installed and used the latest version of traceview.exe from the public WDK.  Here is the info on the traceview file that includes the -i option.   I heard from an external partner the option did not work on an older version of traceview.

[cid:image001.png@01D47CF3.D6750F40]

After installed the SDK and WDK, the file was located in this directory:  C:\Program Files (x86)\Windows Kits\10\Tools\x86

Rob

Robo210, 2018-12-15:
The TraceView command line options are deprecated. Use tracepdb.exe and tracefmt.exe to parse PDBs into  TMF files and parse .etl files into text, respectively.

For what it's worth, the -i parameter to tracepdb.exe requires a list of executable images for which it will download the PDBs (using _NT_SYMBOL_PATH) and extract the TMFs from them. TraceView does not support this option; it will just ignore unknown parameters without any warning message. TraceView, however, will use the events in the trace to automatically identify the PDBs that need to be downloaded. If -i appears to work for you then it is probably because of this information in the trace. Tracepdb.exe has a -i option simply to save you the hassle of first using symchk.exe to download the PDB locally before processing it.

Also, when passing -process to traceview.exe, the options correspond to a very old version of tracefmt.exe, not tracepdb.exe. To do what tracepdb.exe does you'd have to specify the -parsepdb action to TraceView. But again, TraceView's command line options are deprecated and out-of-date compared to the standalone tools.

EliotSeattle, 2018-12-17:
The content has been extensively updated -- thank you, @Robo210 --to take this and more into account.

closed 2018-12-17 by EliotSeattle