MicrosoftDocs/windows-driver-docs #1048: MSISupported on a PCI device that does not support MSI?

tzimmer1, 2018-11-15:
Hello.
Can it be documented what happens if MSI is enabled via registry on a PCI device that does not support MSI?  Is it no-harm?  This potential exists for PCI and PCIe add-in cards that can be plugged into various chipsets whose root ports may or may not support MSI.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 490d0635-430f-0641-73dc-36dc8a8383af
* Version Independent ID: f29cb377-7415-fd93-7194-ecaf24e8b964
* Content: [Enabling Message-Signaled Interrupts in the Registry - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/kernel/enabling-message-signaled-interrupts-in-the-registry)
* Content Source: [windows-driver-docs-pr/kernel/enabling-message-signaled-interrupts-in-the-registry.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/kernel/enabling-message-signaled-interrupts-in-the-registry.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

tzimmer1, 2018-12-14:
Thanks Ted.  I actually did and got a (IMHO) guessing answer from a non-Microsoft contributor.  It would be sweet if I could point the vendors I work with to something more official documentation wise (like here).

tedhudek, 2018-12-17:
Hi @tzimmer1 this is actually ok.

If the hardware truly does not support MSI or MSI-X, this should be indicated in the capability register.

If the capability register indicates that message signaled interrupts are not supported at all, it doesn't matter what the registry says about MSI support.

closed 2018-12-17 by tedhudek

tzimmer1, 2019-01-02:
@tedhudek
Thanks.
Can the documentation page be updated to clearly indicate that enabling MSI via registry on a non-MSI device or root port is OK?