MicrosoftDocs/windows-driver-docs #1052: WDK Version 1809, download link is broken again

PascalTNL, 2018-11-16:
I know there has been an issue with the link before early in October. However that issue has been closed and the download problem occurred in a different way.
This is the result I get when I click on the "Download WDK for Windows 10, version 1809" link.

[image]

What is going on?

EliotSeattle, 2018-11-18:
Hi @PascalTNL. When I click the link, the download begins as it's supposed to. This is on https://docs.microsoft.com/windows-hardware/drivers/download-the-wdk, and the URL for the download for the WDK  for Windows 10, version 1809 is https://go.microsoft.com/fwlink/?linkid=2026156.

closed 2018-11-18 by EliotSeattle

PascalTNL, 2018-11-19:
Hello @EliotSeattle I already used that. I tried different devices and different browsers. But all of them didn't seem to work.

EliotSeattle, 2018-11-28:
The link still seems to work on a number of different machines. Perhaps remove all kits, delete the folders, restart, and try again?
I'm sorry I can't be of more help, as I cannot reproduce your issue.