MicrosoftDocs/windows-driver-docs #1055: Wdmsec.lib is incompatible with POOL_NX_OPTIN opt-in mechanism

vlpast, 2018-11-17:
Hello,

I'm trying to use POOL_NX_OPTIN mechanism to build driver binary that runs in Windows Vista and above. And I had run into problem with IoCreateDeviceSecure function, which actually is a define for WdmlibIoCreateDeviceSecure from WDK's wdmsec.lib static library.

I had to use disassembler and Poolmon utility to understand what's happening. Internally, this function allocates two temporary memory blocks (with PpRb tag) to read some registry settings. Versions of this library from Windows 8+ WDKs use pool type 0x200 = NonPagedPoolNx for these blocks, while older versions use 0x0 = NonPagedPool = NonPagedPoolExecute.

If I link my driver to Windows 7 library, I get warnings from Device Guard Readiness Tool "Incompatible HVCI Kernel Driver Modules found, Reason: execute pool type count: 2".

If I link my driver to Windows 8+ library and load driver into Windows 7 or Vista, these blocks are allocated from unsupported NonPagedPoolNx, though it does not create any visible problem. Poolmon lists them as being allocated from the nonpaged pool.

What should I do?  I feel inclined to link against Windows 8+ library because everything seem to be working as it should in all operating systems that I have tested.

Thank you.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 30ed3997-1721-5972-8777-4dcb0ea6e532
* Version Independent ID: 061a2d78-fffd-890d-cedd-6ba530613082
* Content: [Single Binary Opt-In POOL_NX_OPTIN - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/kernel/single-binary-opt-in-pool-nx-optin)
* Content Source: [windows-driver-docs-pr/kernel/single-binary-opt-in-pool-nx-optin.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/kernel/single-binary-opt-in-pool-nx-optin.md)
* Product: **windows-hardware**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2018-12-14:
Hi @vlpast my apologies for the delay on this one.  Trying to find someone who can answer.  In the meantime, if you haven't already, you can try posting your question on one of the following forums, where other driver developers and Microsoft engineers can be found. The community there is quite responsive.

* [Windows Hardware WDK and Driver Development](https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=wdk)
* [OSR's ntdev List](http://www.osronline.com/showlists.cfm?list=ntdev)

vlpast, 2018-12-14:
Hi Ted,
I've consulted ReactOS kernel source code and have found that these values are used as binary flags actually. So, it seems that passing an unsupported flag in Windows 7/Vista should not have any undesirable effect.
I've released my driver linked to Win8 wdmsec on Dec 11. If I get negative feedback, I'll update the post.
Thank you.

closed 2018-12-17 by tedhudek