MicrosoftDocs/windows-driver-docs #1059: Incorrect IRQL levels for NDIS miniport

I-gor-C, 2018-11-18:
> Every driver function called by NDIS runs at a system-determined IRQL (one of PASSIVE_LEVEL < DISPATCH_LEVEL < DIRQL). For example, a miniport driver's initialization function, halt function, reset function, and sometimes the shutdown function commonly run at PASSIVE_LEVEL. Interrupt code runs at DIRQL, so an NDIS intermediate or protocol driver never runs at DIRQL. All other NDIS driver functions run at IRQL <= DISPATCH_LEVEL.

1. Reset and Shutdown can run at PASSIVE_LEVEL as well
2. "_All other NDIS driver functions run at IRQL <= DISPATCH_LEVEL_" - that is incorrect. For example, OID handler is triggered at PASSIVE_LEVEL, and so is the Pause/Restart handlers
---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 5188dad7-067c-1a97-472a-c63b7e69374c
* Version Independent ID: 5510b9d1-da1a-e014-7fe9-78eacd29f6c8
* Content: [IRQLs in Network Drivers - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/network/irqls-in-network-drivers)
* Content Source: [windows-driver-docs-pr/network/irqls-in-network-drivers.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/network/irqls-in-network-drivers.md)
* Product: **windows-hardware**
* GitHub Login: @duncanmacmichael
* Microsoft Alias: **dumacmic**

duncanmacmichael, 2018-11-19:
Hi @I-gor-C, thanks for pointing this out. As far as I interpret the doc's wording, though, it is technically correct:

1. Reset can run at or below DISPATCH_LEVEL, yes, as is specified on the [callback doc page](https://docs.microsoft.com/windows-hardware/drivers/ddi/content/ndis/nc-ndis-miniport_reset). However, this page says that functions such as reset *commonly* run at PASSIVE_LEVEL, not that they *always* run at PASSIVE_LEVEL. So, there are instances where the system invokes the reset callback at DISPATCH_LEVEL and your code should be able to handle that, but NDIS might also invoke it at PASSIVE_LEVEL.
2. This sentence says that all other NDIS driver functions run *at or below* DISPATCH_LEVEL. So, if an OID handler run at PASSIVE_LEVEL, that is below DISPATCH_LEVEL and fits within what the sentence is saying.

Does this help?

I-gor-C, 2018-11-21:
Hi @duncanmacmichael, thanks for the explanation. I agree, the doc appears to be correct although a bit vague.

closed 2018-11-21 by I-gor-C

duncanmacmichael, 2018-11-26:
No worries @I-gor-C, I was just checking to make sure we were on the same page and that the topic was correct. But, I can certainly clarify it if you think it's still vauge. Besides linking to each of the callback functions, I added this bit to that first sentence you pointed out:

"...commonly run at PASSIVE_LEVEL, although the reset and shutdown functions can be invoked at a higher IRQL if the system requires it."

And I changed the second sentence to say "All other NDIS driver functions run at or below IRQL = DISPATCH_LEVEL" instead of using the "<=" sign, to make it clearer. Hope this helps!