MicrosoftDocs/windows-driver-docs #1063: Links to examples would help

jakeoshins, 2018-11-19:
This page would be so much more useful if at least some of the sections above linked to samples which use them.  Most importantly, I'd like to see the section of an INF which sets the keys.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 7c0c5ca5-0001-4e1c-a8f4-09d9d5329993
* Version Independent ID: 750835f1-e27f-784b-ea88-47ffdc8439a7
* Content: [Registry Values for Debugging WDF Drivers (KMDF and UMDF) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/wdf/registry-values-for-debugging-kmdf-drivers)
* Content Source: [windows-driver-docs-pr/wdf/registry-values-for-debugging-kmdf-drivers.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/wdf/registry-values-for-debugging-kmdf-drivers.md)
* Product: **windows-hardware**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2018-12-14:
Thanks for the feedback @jakeoshins I'll work on updating the page with some examples.  Is there any key in particular for which you'd like a snippet?

jakeoshins, 2018-12-15:
I think that every value listed should mention the INF section that it goes in, or at the very least, the page should be sorted into things that go in .Service and things that go in .HW, etc.  As written, it's not obvious how to construct the INF.

tedhudek, 2019-01-09:
Thanks @jakeoshins for suggesting this one.  I added snippets for VerifierOn, VerboseOn, DbgBreakOnError, and LogPages.

 I'm going to close it out based on our mail with Wei in Dec.

closed 2019-01-09 by tedhudek