MicrosoftDocs/windows-driver-docs #1068: Attestation Signing DOES Work for Server

PeterGV, 2018-11-20:
The following comment is blatantly wrong, and is causing folks in the community immeasurable pain and suffering:

&gt;To load on Windows Server 2016, drivers must pass HLK tests and be signed by the Hardware 
&gt;dashboard portal. Windows Server does not load attestation signed drivers.

I have spoken to the PM at MSFT who is responsible for Server (MSFT friends, contact me offline if you want the name).  He has verified the above is not correct, and that the present time there are no concrete plans for requiring every driver on Server to pass the HLKs.

Please fix this.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 1eb19d33-8f4c-f379-b645-938a746c5615
* Version Independent ID: 8efb9fb4-1e9a-1a8a-994b-450495449e15
* Content: [Driver Signing Policy - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/kernel-mode-code-signing-policy--windows-vista-and-later-)
* Content Source: [windows-driver-docs-pr/install/kernel-mode-code-signing-policy--windows-vista-and-later-.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/kernel-mode-code-signing-policy--windows-vista-and-later-.md)
* Product: **windows-hardware**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2018-11-20:
Following up with Peter offline

tedhudek, 2018-11-21:
This change is live

rpetti, 2019-07-22:
This change has not appear to have gone live. All of the documentation on signing still states that attestation does not work for Server versions.

For example:
https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/get-drivers-signed-by-microsoft-for-multiple-windows-versions

> But it’s important to note that a submission signed this way will not work on Windows Server 2016.

We would very much like some clarity on this. Can or can't attestation be used to sign drivers on Windows Server?

tedhudek, 2019-07-22:
Hi @rpetti, I'm pretty sure you can install an Attestation driver on a Server 2016+ system.  It looks like I was referring to 77ff596aaa in my above statement, which did remove that language from the page this issue references.  I'll keep poking around for more info here but wanted to get that out to you

tedhudek, 2019-07-24:
Confirmed. Just pushed an update to https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/get-drivers-signed-by-microsoft-for-multiple-windows-versions to remove that sentence.  Give it 20 min for the change to reach the live site

czarau, 2019-12-04:
Still referred to here....
https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/attestation-signing-a-kernel-driver-for-public-release

PeterGV, 2019-12-04:
It took me a couple of reads, but the docs ARE still incorrect (incorrect phrase highlighted in bold):
---
An attestation signed driver will only work for Windows 10. It will not work for other versions of Windows, such as Windows 8.1, Windows 7, **or versions of Windows Server later than Windows Server 2016**.
---

EliotSeattle, 2019-12-04:
@tedhudek

tedhudek, 2019-12-04:
Just pushed the change, should be live soon.  Appreciate the heads-up.

closed 2019-12-04 by tedhudek