MicrosoftDocs/windows-driver-docs #1069: Does the /status feature still work with latest releases of symproxy?

ryan-roberts, 2018-11-20:
Doing so with our installation of v10.0.17763.132 (from the 20181023 Win10 SDK) just returns a 404.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: ded21cca-26e1-591d-67a7-9b533d2cb659
* Version Independent ID: 05263644-c5b2-64b7-fb55-3c0dc37dc5df
* Content: [Checking and Updating Status - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/checking-and-updating-status)
* Content Source: [windows-driver-docs-pr/debugger/checking-and-updating-status.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/checking-and-updating-status.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2018-11-28:
@ryan-roberts - I'm checking with our dev team to see if this is still supported or if there is a missing configuration step here.

DOMARS, 2019-01-04:
@ryan-roberts - As it turns out this feature was removed from this product. So I will update the docs to longer mention that this is available. Thanks for pointing out this error in our documentation.

closed 2019-01-04 by DOMARS