MicrosoftDocs/windows-driver-docs #1073: a Typo?

airwolf2026, 2018-11-23:
hardwareid root\\sysvad\_TabletAudioSample
should be root\sysvad_TabletAudioSample

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 9226c2e0-2c63-788b-5164-4856f66cc12b
* Version Independent ID: ba393947-45d1-935d-aba2-b776e163915b
* Content: [Debug Drivers - Step-by-Step Lab (Sysvad Kernel Mode) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/debug-universal-drivers--kernel-mode-)
* Content Source: [windows-driver-docs-pr/debugger/debug-universal-drivers--kernel-mode-.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/debug-universal-drivers--kernel-mode-.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2018-11-27:
@airwolf2026 - Thanks for calling out an issue here. It looks like as we were converting the text extra slashes were added in, that are no longer needed. The updated topic should be live in a day or so.

DOMARS, 2018-11-29:
@airwolf2026 - The fix for the extra slashes is live, so closing this issue.

closed 2018-11-29 by DOMARS