MicrosoftDocs/windows-driver-docs #1075: Filter driver missing in illustrates

agowa338, 2018-11-24:
The Filter driver is not shown in the illustration.

Also this illustration may should also be copied to or at least linked to from:
- NDIS driver types - Windows drivers | Microsoft Docs https://docs.microsoft.com/en-us/windows-hardware/drivers/network/ndis-drivers
- Introduction to Network Drivers - Windows drivers | Microsoft Docs https://docs.microsoft.com/en-us/windows-hardware/drivers/network/introduction-to-network-drivers

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: e0510b76-a16d-f571-82a6-7b94fcb0d153
* Version Independent ID: 1f2ed947-ff5b-7eec-5060-404baaaff33e
* Content: [Intermediate drivers - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/network/ndis-intermediate-drivers)
* Content Source: [windows-driver-docs-pr/network/ndis-intermediate-drivers.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/network/ndis-intermediate-drivers.md)
* Product: **windows-hardware**
* GitHub Login: @duncanmacmichael
* Microsoft Alias: **dumacmic**

duncanmacmichael, 2018-11-26:
Hi @agowa338, thanks for the thoughtful feedback. Here's my take, which we can use as a starting point for discussion if you'd like. 

For your first point: after examining this page, the pages to which you linked, and the [Filter driver](https://docs.microsoft.com/windows-hardware/drivers/network/ndis-filter-drivers) page, I think the diagrams are appropriate as-is. Intermediate drivers sit between protocol drivers and miniport drivers and act as both, hence including all three in the illustration for intermediate drivers. Filter drivers are a lighter-weight alternative to intermediates and are independent of them. Technically, either page's illustration could include the other, but it isn't necessary to understand how each one works. 

For your second point, perhaps you are looking for more of an overall diagram that shows all the driver types, like on the [NDIS Driver Stack](https://docs.microsoft.com/windows-hardware/drivers/network/ndis-driver-stack) topic. But, I can certainly link to there from these individual pages if you think that would help.

What do you think?

agowa338, 2018-11-30:
LGTM.

duncanmacmichael, 2018-11-30:
@agowa338 Great, thanks for the feedback! I'll go ahead and close this issue, then.

closed 2018-11-30 by duncanmacmichael