MicrosoftDocs/windows-driver-docs #1077: Windows Driver Kit (WDK) Installation Error: “setup cannot locate windows debugging wdk integration-x86_en-us.msi”

ydididodat, 2018-11-25:
When I try to install WDK from https://docs.microsoft.com/en-us/windows-hardware/drivers/download-the-wdk It gets stuck to 0% for a long time until it finally displays the error message saying "setup cannot locate windows debugging wdk integration-x86_en-us.msi".

I'm using Windows 10 64bit v1809. I have installed Visual Studio Community 2017 through the Visual Installer. I also installed the Windows SDK as an optional component which was selected by default. 

I also see in my start menu that I have under recently installed: WinDbg (X86).

I have tried creating a VirtualBox VM with a fresh install of Windows 10 64bit v1809, so it's exactly the same as my host OS. But if I try to install WDK there, then I receive the exact same problem. 

I've also tried installing the previous WDK version but it didn't make any difference.

EliotSeattle, 2018-11-26:
Hi @ydididodat.

I just tried the install on several machines, and it works just fine. It opens the installer and is ready to go in about 10-15 seconds. Maybe try uninstalling any previous kits that you have?

Sorry I can't be of more help. I just can't reproduce this.

@EliotSeattle

ydididodat, 2018-12-07:
@EliotSeattle That's strange. However, A Microsoft employee said they were able to reproduce the error and would look into it. You can take a look here: https://social.msdn.microsoft.com/Forums/en-US/d414ccec-16d6-4cf8-829b-6fb6114d8bfc/setup-cannot-locate-windows-debugging-wdk-integrationx86enusmsi?forum=wdk

What do you think about that?

EliotSeattle, 2018-12-07:
I think it's great that our engineers are better at reproducing problems than our content managers! I've reached out to Anish, and will update this once a solution is delivered.
Thanks for following up with me. I appreciate it.

EliotSeattle, 2019-03-06:
I'm sorry to say this, but I am unable to contact the person who reproduced the error, and nobody else has been able to do so.

zieserl, 2021-06-16:
Hello, i have this same issue now, running the installer in an Hyper-V emulated system environment.

tedhudek, 2021-06-16:
Thanks for the report, Luca.  What OS are you running in the Hyper-V environment?

cc @Cymoki

zieserl, 2021-06-18:
> 
> 
> Thanks for the report, Luca. What OS are you running in the Hyper-V environment?
> 
> cc @Cymoki

Windows 10 Pro 21H1 64bit

tedhudek, 2021-06-18:
Thanks @zieserl, I'm investigating.  In the meantime, you could the EWDK.  I haven't tried it but I believe you can install everything as an .iso - might be easier.

Have you tried posting to the current driver forums?  Looks like social got replaced by https://docs.microsoft.com/en-us/answers/products/.  There is also http://www.osronline.com/showlists.cfm?list=ntdev.

tedhudek, 2021-06-18:
Circling back with questions from the dev team --

Do you know version of WDK is being installed and if the SDK was installed prior to that? Have you tried uninstalling the kit completely and running again?

tedhudek, 2021-06-23:
Hi Luca, I'm going to close this one out since I don't have any more info for you.  The feature team was not familiar with this error.  If you see it again please try the above.  Thanks for contacting us, and best of luck!

closed 2021-06-23 by tedhudek