MicrosoftDocs/windows-driver-docs #1078: The latest sample isn't completed.

wayyt, 2018-11-26:
The latest sample isn't completed.

// Example of an event
 [WMI, guid("xxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx"),
 Dynamic, Provider("WMIProv"),
 locale("MS\\0x409"),
 Description("Serial Send Event"),
 WmiExpense(1)]
class Vendor_SerialSendEvent : WMIEvent
{
// Required items
    [key, read] 
      string InstanceName;
    [read]
      boolean Active;

Need to post the rest content in here.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: a71ac393-9940-cb7e-e254-27839dbed116
* Version Independent ID: e961a9a1-a9f0-283b-19d6-dacb738a77b5
* Content: [WMI Class Examples - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/kernel/wmi-class-examples)
* Content Source: [windows-driver-docs-pr/kernel/wmi-class-examples.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/kernel/wmi-class-examples.md)
* Product: **windows-hardware**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2018-12-14:
Hi @wayyt sorry for the delay.  Trying to find someone who can answer your question.  What is it that you are trying to do?

tedhudek, 2018-12-19:
Hi @wayyt you are correct!  I've updated the sample code.  Change will appear on the doc page in a few minutes, but here's the commit: 7da1ae19da23955a7ca2a245b218d68d65df59a5.  Thanks again for the feedback!

closed 2018-12-19 by tedhudek