MicrosoftDocs/windows-driver-docs #1079: Inaccuracies

vo-va, 2018-11-26:
All notes related only to Window 10 Pro 1803, build 17134.407

## Windows 10 partly support Passpoint R2, including Online Signup.

From my experience there is some inaccuracies in Passpoint R2 implementation.

Passpoint R2 specification, chapter "2. Hotspot 2.0 Device, Operator and Service Provider requirements", define that mobile device should support these authentication methods.

| Credential  | Type EAP Method |
| ------------- | ------------- |
| Certificate  | EAP-TLS  |
| SIM/USIM  | EAP-SIM,  EAP-AKA  |
| Username/Password (with server side certificates)  | EAP-TTLS with MSCHAPv2  |

Windows uses this file `c:\Windows\System32\TransformPPSToWlan.xslt` to transform Mgmt tree to Windows wlan profile file. It has next statement

&gt;Currently transformation supports following authentication outer and inner methods
&gt;
&gt;      1) EapTTLS with PAP
&gt;      2) EapTTLS with MSCHAPv2 (default if nothing specified)

I don't now about EAP-SIM/EAP-AKA supporting status, but at least Windows 10 does not support EAP-TLS.

## Windows uses wrong id in Mgmt tree
Passpoint R2 specification, chapter "9.2.1 Graphical representation" states next

&gt;This extension shall be uniquely identified by using following path in the DevDetail subtree: `./DevDetail/Ext/org.wi-fi/Wi-Fi`.

From my experience Windows 10 uses `./DevDetail/Ext/org-wifi/Wi-Fi`.

## Windows does not understand PolicyUpdate, SubscriptionUpdate credentials node.

It uses this file `c:\Windows\System32\TransformPPSToWlanCredentials.xslt` to extract credentials. It will select first credentials from mgmt tree.

Please apply this changes:

```diff
--- a/TransformPPSToWlanCredentials.xslt
+++ b/TransformPPSToWlanCredentials.xslt
@@ -7,7 +7,7 @@
     &lt;xsl:template name="PasspointCredentials" match="/"&gt;
       &lt;!-- EapTtls based credentials --&gt;
       &lt;xsl:call-template name="EapTtlsCredentials"&gt;
-        &lt;xsl:with-param name="credsNode" select="//Node[NodeName='UsernamePassword']"/&gt;
+        &lt;xsl:with-param name="credsNode" select="//Node[NodeName='Credential']/Node[NodeName='UsernamePassword']"/&gt;^M
       &lt;/xsl:call-template&gt;
     &lt;/xsl:template&gt;
``` 

And please add logging from `wifitask.exe` to Windows event journals. It's hard to debug binary file without any error/debug log messages. 

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 0e9ccdc2-b2e2-68cd-2661-f7f3db1ad6fd
* Version Independent ID: 48ba8f38-233f-010f-37bb-d3d17b63e12f
* Content: [Hotspot 2.0 - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/mobilebroadband/hotspot-20)
* Content Source: [windows-driver-docs-pr/mobilebroadband/hotspot-20.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/mobilebroadband/hotspot-20.md)
* Product: **windows-hardware**
* GitHub Login: @duncanmacmichael
* Microsoft Alias: **dumacmic**

duncanmacmichael, 2018-11-27:
Hi @vo-va, thanks so much for the detailed feedback. To me, though, this appears to be product feedback rather than documentation feedback. Let me see what the best option would be for getting you to the right place here.

duncanmacmichael, 2019-02-04:
@vo-va Hi again, sorry for the delay. I wasn't able to get a response about this issue as far as it's related to documentation, so due to lack of activity I'll go ahead and close this issue. We do appreciate the feedback and always welcome it, but perhaps this particular feedback would be better directed to the devs and PMs who work on Windows rather than the docs. I'd recommend taking this and using the Product Feedback button on the bottom of the page to submit it, which will direct it to the Windows Developer voice site and get it to the right people. Thanks again for your interest in the docs.

closed 2019-02-04 by duncanmacmichael

deepaksehrawat, 2019-12-17:
Any update on this issue? We are also facing problem while testing Passpoint R2 feature with windows 10.

I am trying to provision Hotspot 2.0 (Passpoint) profile on my windows-10 laptop using Online SignUp Server (OSU) using SOAP based provisioning.

- Windows is able to successfully discover and select an OSU SSID

- Successfully establishes a TLS session with OSU Server

- Sends "SOAP: sppPostDevData (requestReason, redirectURI, DevInfo + DevDetail MOs)" request

- Successfully receives "SOAP: sppPostDevDataResponse (sessionID, sppStatus, exec: launchBrowserToURI)" response and opens up Web Browser

- Performs registration

- Sends "SOAP: sppPostDevData (sessionID, requestReason, DevInfo+DevDetail MOs)" request to OSU Server

- Receives "SOAP: sppPostDevDataResponse (sessionID, sppStatus, addMO: PerProviderSubscription MO)" response from OSU Server

- BUT Sends back "SOAP: sppUpdateResponse (sessionID, sppStatus)" message with error code as "Error Occured"

We tried to debug it via Event Logs but could not find any detailed reason code as why it is returning this error. Our SOAP response is absolutely inline with Hotspot2.0 release 2.0 specification (8.4.2.1 Provisioning username and password credentials).

Can anyone help me with any pointer to debug this issue? If we can get detailed log at what point it is failing (while processing OSU response) on windows laptop. Which XML parameter it is not able to decode/rejecting? 

We can even attach the request / response logs (SOAP) if required.

https://answers.microsoft.com/en-us/windows/forum/all/wifi-hotspot-20-passpoint-related-issue/f32b6ca7-c6c9-46ac-b921-e9624cf04b8d

billlattimer, 2020-01-06:
@deepaksehrawat this issue was closed in Feb 2019. The issue you're facing may be implementation-specific; you may want to direct this towards online developer communities for the OSU and for SOAP.

vo-va, 2022-08-09:
@deepaksehrawat can you shared your payload for sppPostDevDataResponse that returns error? Maybe I can help.