MicrosoftDocs/windows-driver-docs #1080: Type for the sympath command

buckleyGI, 2018-11-26:
Typo I guess
Search for ".sympath srv*."
That commands ends with a point "*" but shouldn't

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 142adea7-b7b8-4a5a-5e19-83a202ab04b2
* Version Independent ID: 489c0c22-908c-aa93-520c-836c941b5b4e
* Content: [Getting Started with WinDbg (User-Mode) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/getting-started-with-windbg)
* Content Source: [windows-driver-docs-pr/debugger/getting-started-with-windbg.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/getting-started-with-windbg.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2018-11-28:
@buckleyGI - Looks like that period was for the end of sentence, but it does make things confusing, so it will be removed.

DOMARS, 2018-12-10:
@buckleyGI - The period is removed. Thanks for pointing this out.

closed 2018-12-10 by DOMARS