MicrosoftDocs/windows-driver-docs #1086: Whats the registry key is located

RAJU2529, 2018-11-27:
I am using windows 10 enterprise with device gaurd by gpo ,  How  to  find  the regsitry  key  for 
MOR byte is set to 0x11,

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: f1b5ea68-0be9-fb7f-a044-08790b6a1da2
* Version Independent ID: e67e1c77-c6ac-db82-277d-7b8126e3b1eb
* Content: [Memory Overwrite Request Control (MOR) LOCK version 2 - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/bringup/memory-overwrite-request-control-lock-version-2)
* Content Source: [windows-driver-docs-pr/bringup/memory-overwrite-request-control-lock-version-2.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/bringup/memory-overwrite-request-control-lock-version-2.md)
* Product: **windows-hardware**
* GitHub Login: @barrygolden
* Microsoft Alias: **bagold**

matthewfcarlson, 2019-10-02:
As far as I know, this is a UEFI variable and won't be accessible from the register. It is instead stored in the firmware itself (depending on the platform, the implementation may vary). But the variable can be read via the UEFI variable tool. https://docs.microsoft.com/en-us/windows/win32/sysinfo/access-uefi-firmware-variables-from-a-universal-windows-app

Or you can read Vincent Zimmer's blog (a very knowledgable person in the UEFI community) https://vzimmer.blogspot.com/2012/12/accessing-uefi-form-operating-system.html

barrygolden, 2020-12-18:
Thanks for reaching out to the Windows Driver Developer Documentation team. 

You may have success posting your question in the following forums, where other driver developers and Microsoft engineers can be found. The community there is quite responsive.

[MSDN Windows Hardware WDK and Driver Development forum](https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=wdk)

[NTDEV forum at OSR Online](https://www.osronline.com/showlists.cfm?list=ntdev)

closed 2020-12-18 by barrygolden