MicrosoftDocs/windows-driver-docs #1097: Unclear/inconsistent documentation of network device Surprise removal

I-gor-C, 2018-11-29:
From the doc:
> 5. If the miniport driver was successfully initialized, NDIS calls the MiniportDevicePnPEventNotify function with an event code of NdisDevicePnPEventSurpriseRemoved. The miniport driver should note that the device has been physically removed. If the miniport driver is an NDIS-WDM driver, it should cancel any pending IRPs that it sent down to the underlying bus driver. If the miniport driver was not successfully initialized, processing continues.
> 6. NDIS performs the following steps:
>     a. It pauses all protocol drivers that are bound to the NIC.
>     b. It pauses all filter drivers that are attached to the NIC.
>     c. It pauses the miniport driver for the NIC.

I suppose the _**processing continues**_ in par.5 means that the algorithm proceeds to the next step (6)? If that's true then 6.a describes an invalid transition in miniport SM. If initialization has failed the miniport returns to the Halted state. Triggering Pause from Halted state isn't a valid transition (per https://docs.microsoft.com/en-us/windows-hardware/drivers/network/adapter-states-of-a-miniport-driver ).

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 619ff26a-1b82-74fe-db0a-b0b17e370730
* Version Independent ID: ba894464-1bd8-7d9e-6f7c-b6ccd21a4d9f
* Content: [Processing the surprise removal of a NIC (Windows 7 and later) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/network/processing-the-surprise-removal-of-a-nic--windows-7-and-later-versions-)
* Content Source: [windows-driver-docs-pr/network/processing-the-surprise-removal-of-a-nic--windows-7-and-later-versions-.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/network/processing-the-surprise-removal-of-a-nic--windows-7-and-later-versions-.md)
* Product: **windows-hardware**
* GitHub Login: @duncanmacmichael
* Microsoft Alias: **dumacmic**

duncanmacmichael, 2018-11-29:
Hmm, interesting question @I-gor-C. I'm not sure, either, so let me see what I can find out from the feature team to clarify what the doc is saying.

jtippet, 2018-11-30:
I helped draft that documentation in 2011, and yet I honestly cannot remember what's going on with the "processing continues" text.  It's vacuously true -- if the driver fails MiniportInitializeEx, then there are no filters or protocols to pause or unbind, so step 6 is a no-op.

I'd just ignore "If the miniport driver was successfully initialized" and "If the miniport driver was not successfully initialized, processing continues.".  They're not helpful.

duncanmacmichael, 2018-11-30:
Great, thanks @jtippet! That makes sense. I've merged your PR so those changes should be live later today.

closed 2018-11-30 by duncanmacmichael