MicrosoftDocs/windows-driver-docs #1098: Values for DriverPackageType not specified

BrianCatlin, 2018-11-29:
Where are we supposed to find the legal values for DriverPackageType?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 89e65b9c-b342-6df7-e878-4b88a9590363
* Version Independent ID: 8f0d7eb7-cd13-2769-f20d-296ecc96d417
* Content: [INF Version Section - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/inf-version-section)
* Content Source: [windows-driver-docs-pr/install/inf-version-section.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/inf-version-section.md)
* Product: **windows-hardware**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2018-12-14:
Hi @BrianCatlin sorry for the delay.  Trying to find someone who can help, please stay tuned.

tedhudek, 2018-12-17:
Turns out DriverPackageType is deprecated because it was used by DIFx.