MicrosoftDocs/windows-driver-docs #1099: Do I need this or not?

BrianCatlin, 2018-11-29:
The first line reads "This section allows users to request additional signatures as required by certain certification scenarios". So, what are those scenarios? Where can I find information on them (specifically, not some vague "look it up in the HCK docs")? I'm writing an ELAM driver, so is this section required, or not?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: ba708169-9e8f-f6fc-c0c8-8ba44874d96a
* Version Independent ID: b9304f44-194a-a275-38f4-9d3d4b3d32cb
* Content: [INF SignatureAttributes Section - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/inf-signatureattributes-section)
* Content Source: [windows-driver-docs-pr/install/inf-signatureattributes-section.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/inf-signatureattributes-section.md)
* Product: **windows-hardware**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2018-12-14:
Reaching out to ELAM folks, will keep you posted

tedhudek, 2018-12-15:
Hi @BrianCatlin the answer is, yes it is.  I'll update the docs as soon as I can but wanted to get the info to you.

BrianCatlin, 2018-12-15:
Thank you for pursuing this @tedhudek !

tedhudek, 2018-12-17:
Updated the page. You can see the updates for this issue and #1100 in 1bdca8da2

closed 2018-12-17 by tedhudek