MicrosoftDocs/windows-driver-docs #1101: Warning 1205 missing

BrianCatlin, 2018-11-29:
I'm getting warning 1205 "Section [Driver_files] referenced from DelFiles and CopyFiles directive". Why isn't it listed here? Why is this a warning?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 48275f87-4086-1ca5-a190-c741f29ad348
* Version Independent ID: 5481885d-3063-527b-a85e-6a898dbcb990
* Content: [INF Validation Errors and Warnings - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/inf-validation-errors-and-warnings#err-11xx)
* Content Source: [windows-driver-docs-pr/devtest/inf-validation-errors-and-warnings.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/inf-validation-errors-and-warnings.md)
* Product: **windows-hardware**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

EliotSeattle, 2018-11-30:
Thanks for pointing this out, @BrianCatlin. I've added the following to the table:

**1205: Section [Driver_files] referenced from [Directive1] and [Directive2] directive**
This warning is generated whenever two different directives point to the same section.

I just created the pull request, so it should be live shortly.

closed 2018-11-30 by EliotSeattle

BrianCatlin, 2018-12-01:
You should point out that multiple directives pointing to the same section is frequently an error, but that is not always the case (false positive)