MicrosoftDocs/windows-driver-docs #1104: Troubleshooting and resolution steps needed.

LuigiBruno, 2018-11-29:
Add troubleshooting and resolution steps for this bug check.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 453b231e-8201-9fcc-34ac-32f4ee6fbb82
* Version Independent ID: 6c20f054-3d42-e020-69a1-843e4a87d26a
* Content: [Bug Check 0xF4 CRITICAL_OBJECT_TERMINATION - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0xf4--critical-object-termination)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0xf4--critical-object-termination.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0xf4--critical-object-termination.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-01-17:
@LuigiBruno - Thanks for this issue. As I have mentioned before, we use a process to prioritize the most frequent bug checks out of the hundreds that are currently defined. With our limited documentation team resources, it is likely that we will only be able to do this work for a small sub set of all of the possible stop codes.
Right now I'm working on the new stop codes that we have not documented at all.
Thank you for your interest in improving our stop code docs.

DOMARS, 2019-09-11:
@LuigiBruno - I have opened an internal work item to track this work, so closing this issue for now.

As I have mentioned before, we use a process to prioritize the most frequent bug checks out of the hundreds that are currently defined. With our limited documentation team resources, it is likely that we will only be able to do this work for a small sub set of all of the possible stop codes.

Thank you for your interest in improving our stop code docs.

closed 2019-09-11 by DOMARS