MicrosoftDocs/windows-driver-docs #1107: SignatureAttributes requirement needs to be on the ELAM driver requirements page!

BrianCatlin, 2018-11-30:
Why isn't the note (item #1) about SignatureAttributes on the ELAM drive requirements page where it belongs? Having it only on this page doesn't make sense because the INF file is written early in the development process, and isn't changed once it gets the driver loaded

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: f969a593-1e57-9e24-5fe1-fc1d372fd571
* Version Independent ID: 76a0f061-edc8-76cb-dc82-3b3f8532e06e
* Content: [ELAM Prerequisites - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/elam-driver-submission)
* Content Source: [windows-driver-docs-pr/install/elam-driver-submission.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/elam-driver-submission.md)
* Product: **windows-hardware**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2018-12-17:
So is it in https://docs.microsoft.com/en-us/windows-hardware/drivers/install/elam-driver-submission, does that work?

BrianCatlin, 2018-12-17:
No. All that stuff should either be in one place, or there should be a note to look at the submission page for additional requirements, for the reason that I specified in my original comment, above

tedhudek, 2018-12-19:
Done.  I did a little restructuring of that whole doc node.  Thanks again for the feedback!

closed 2018-12-19 by tedhudek