MicrosoftDocs/windows-driver-docs #1108: Need info on creating a signing certificate

BrianCatlin, 2018-11-30:
The unique code signing requirements of an ELAM driver with a protected process are not sufficiently documented. Please provide a PowerShell script that uses New-SelfSignedCertificate (because MakeCert,exe is officially deprecated) to create the necessary certificates 

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 1b2d3984-d4fc-5e9a-1dfe-b5593e308c76
* Version Independent ID: ad920b64-d6dd-46b8-ffab-c8bad9dc9de3
* Content: [ELAM Prerequisites - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/elam-prerequisites)
* Content Source: [windows-driver-docs-pr/install/elam-prerequisites.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/elam-prerequisites.md)
* Product: **windows-hardware**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2018-12-18:
Hey @BrianCatlin here's what I got back from the PM:

Is this specific to testing scenarios?  There is no scenario where a partner needs to make their own cert for a shipping ELAM driver (and for AM-PPL they can use any cert they want to sign the service, the hash just needs to be present in the ELAM driver resource section).

BrianCatlin, 2018-12-18:
What about for testing? A regular test-signing certificate will not work. After about a day of trial and error, I finally figured it out and I wrote the following script to generate an ELAM certificate. You should extract the salient parts and put them in the doc

```
#
# Generate a code signing certificate with the Early-Launch Anti-Malware (ELAM) EKU for the  
# specified project, and save the public and private keys in files. 
#
# The private key will be in a file named "<project name>-Private.pfx", and the password for 
# the key will be the project name (case sensitive). The private key should be used on 
# development machines for signing executables (Visual Studio project properties->Driver Signing
# ->Test Certificate, and should be installed using the script Install-TestCodeSigningCertificate.ps1
#
# The public key will be in a file named "<project name>-Public.cer" and should be installed on 
# machines running the driver using the script Trust-TestCertificate.ps1
#
# Brian Catlin
#

[CmdLetBinding ()]
param
    (
    [Parameter (Mandatory=$false, HelpMessage="Project name")]
    [ValidateNotNullOrEmpty ()]
    [string] $Name
    )

#
# The files to create
#

$private_key_file = "$Name-Private.pfx"
$public_key_file = "$Name-Public.cer"

#
# Ensure that we don't over-write existing certificates
#

if (!(Test-Path -Path $public_key_file) -and !(Test-Path -Path $private_key_file))
    {

    #
    # Generate the certificate
    #

    $cert = New-SelfSignedCertificate -CertStoreLocation cert:\currentuser\My `
        -Subject "CN=$Name Development - For testing only" `
        -KeyAlgorithm RSA `
        -KeyLength 2048 `
        -Provider "Microsoft Enhanced RSA and AES Cryptographic Provider" `
        -KeyExportPolicy Exportable `
        -KeyUsage DigitalSignature `
        -Type CodeSigningCert `
        -TextExtension @("2.5.29.37={text}1.3.6.1.5.5.7.3.3, 1.3.6.1.4.1.311.61.4.1")

    #
    # Export the private key
    #

    $secure_pw = $Name | ConvertTo-SecureString -AsPlainText -Force
    Export-PfxCertificate -Cert $cert -Password $secure_pw -FilePath $private_key_file

    #
    # Export the public key
    #

    Export-Certificate -Cert $cert -Type CERT -NoClobber -FilePath $public_key_file
    }
else
    {
    Write-Error "Key file already exists"
    }

```

tedhudek, 2019-02-14:
Hi @BrianCatlin sorry to take a while on this one.  Turns out we can't drop your code in the doc without testing, which we don't have bandwidth to do currently.  However, even when closed, this issue will remain linked at the bottom of the page on docs.ms, so people will be able to see your proposed code without leaving the page.  Best I can do at the moment!

closed 2019-02-14 by tedhudek