MicrosoftDocs/windows-driver-docs #1118: i received  warnings when i entered command

RAJU2529, 2018-12-01:
Today i verified the .inf  file , whether as universal driver .  notably  i  have  universal driver "iigd_dch.inf"

cd /d C:\Program Files (x86)\Windows Kits\10\tools\x64\
infverif.exe C:\oem1\iigd_dch.inf /V /U /Osver NTamd64.6.0 

but i have received  some errors 

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 2eed9920-f07e-8e74-4ad1-d86803041217
* Version Independent ID: 1bd1e894-d45d-c6f8-b7ba-947320f76cd5
* Content: [Running InfVerif from the command line - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/running-infverif-from-the-command-line)
* Content Source: [windows-driver-docs-pr/devtest/running-infverif-from-the-command-line.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/running-infverif-from-the-command-line.md)
* Product: **windows-hardware**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

EliotSeattle, 2018-12-06:
Hi @RAJU2529.
 Unfortunately, we don’t have the resources to offer individual technical or coding advice to the broad array of customers who use the driver documentation.
You may have success posting your question in the following forums, where other driver developers and Microsoft  engineers can be found. The communities there are quite responsive.
- [Windows Hardware WDK and driver development forum](https://social.msdn.microsoft.com/Forums/windowshardware/en-US/home?forum=wdk)
- [OSR NTDEV forum](http://www.osronline.com/cf.cfm?PageURL=showlists.cfm?list=NTDEV)
 
Again, thanks for reaching out. Best of luck with this issue.

closed 2018-12-06 by EliotSeattle

RAJU2529, 2018-12-06:
its Not coding advice