MicrosoftDocs/windows-driver-docs #1127: The stornvme driver

mfstanton, 2018-12-08:
I am fighting a problem with stornvme error recovery. I cannot find any information that relates specifically to that driver nor can I find anything that provides hints about what it does during error recovery.  It gets an error on the device and resets it then it takes a reboot to recover the device. Not a healthy or robust environment 

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: c84d98e4-5ff2-205a-620f-5e4cb2ecc398
* Version Independent ID: f540c12c-446d-dac2-0976-82179bc405fa
* Content: [Storport Miniport Drivers - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/storage/storport-miniport-drivers)
* Content Source: [windows-driver-docs-pr/storage/storport-miniport-drivers.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/storage/storport-miniport-drivers.md)
* Product: **windows-hardware**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

EliotSeattle, 2019-04-26:
@lorihollasch, can you please look at this?

lorihollasch, 2019-06-14:
@mfstanton, sorry for the slow response to this - hope you have moved past your error situation by now. The details of how proprietary storage miniport drivers work aren't documented - this includes stornvme. I did update the [Storport Event Log Extensions](https://docs.microsoft.com/en-us/windows-hardware/drivers/storage/storport-event-log-extensions) page to point out where Storport logs events. I don't know that stornvme logs anything of use for you, but it was useful information regardless.

closed 2019-06-14 by lorihollasch

mfstanton, 2019-06-14:
Well I have stopped using them on one machine. The design has a major flaw in it and there is a failure on timeouts to correctly process the reset of a PCIe Bus that is bifurcated by a bridge. You can’t just document your way out of this problem. Reliability  is something people count on.  I spent a lot of time fixing similar issues for my device driver on an enterprise Linux system  I would be happy to discuss it with someone and give them the benefit of my experience in that realm.  Enterprise customers do not take “sorry that is the way things are” as an excuse for system outages or data corruption because the reset of a bus is not handled adequately.  There  is also a matter of a device timeout being handled without regard for the size of the transfer and the geometry of the target devices.

lorihollasch, 2019-06-21:
@mfstanton, I have passed your feedback on to the dev team and asked them to reach out to you when they are able. Thanks for your offer to share your expertise!