MicrosoftDocs/windows-driver-docs #1134: commitStatus field of Submission resource is undocumented

winter3angle, 2018-12-11:
Subj, there is no documentation on possible values for commitStatus enumeration.
For now I know that there are at least `commitPending` and `commitReceived` members exist.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: a6788ab0-e8c3-b5c2-c127-6528e4b4231d
* Version Independent ID: 4a618589-7215-89cb-8bc1-330ba9131ddd
* Content: [Get product data - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/get-product-data#submission-resource)
* Content Source: [windows-driver-docs-pr/dashboard/get-product-data.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/get-product-data.md)
* Product: **windows-hardware**
* GitHub Login: @balapv
* Microsoft Alias: **balapv**

winter3angle, 2018-12-11:
Also `commitStarted`, it's mentioned on the page on product submission committing

tedhudek, 2020-11-03:
Related to #839

tedhudek, 2020-11-09:
Thanks for the feedback and apologies for the delay.  I've updated these values in 7f72cade9.

closed 2020-11-09 by tedhudek