MicrosoftDocs/windows-driver-docs #1137: Insufficient details

vijairaj, 2018-12-12:
Please add a link to (https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/controlling-exceptions-and-events) which has better details on `Execution option` and `Continue option`

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 80d9f5e1-6e9f-fc6b-8b25-510282f182d6
* Version Independent ID: 7ab82b23-e425-bece-3756-76f09c6f79e6
* Content: [Debug Event Filters - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/debug---event-filters)
* Content Source: [windows-driver-docs-pr/debugger/debug---event-filters.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/debug---event-filters.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-01-18:
@vijairaj - Not sure that I follow. At the bottom of the topic under "Additional Information" we already have this text:
*For more information about break status and handling status, all event codes, the default status for all events, and other methods of controlling this status, see Controlling Exceptions and Events."*

DOMARS, 2019-06-07:
@vijairaj - As the page you mentioned is already linked from this topic, I will close this issue.  Thanks for reaching out to the Windows Driver Developer Documentation team.

closed 2019-06-07 by DOMARS