MicrosoftDocs/windows-driver-docs #1138: DriverMetadata member missing in Downloads resource description

winter3angle, 2018-12-12:
There is `driverMetadata` download type also exists, it could be seen eg. when your submission has just passed over `finalizeIngestion` workflow step. At least I encountered it in that case.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 1c20769b-d969-a51a-4eb3-f9cc27875f28
* Version Independent ID: 853fe2a9-d654-98a3-7635-c30a450cdbdc
* Content: [Get product data - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/get-product-data#submission-resource)
* Content Source: [windows-driver-docs-pr/dashboard/get-product-data.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/get-product-data.md)
* Product: **windows-hardware**
* GitHub Login: @balapv
* Microsoft Alias: **balapv**

winter3angle, 2018-12-12:
Probably it was confused with Rel member of Link resource. Look a these:
 * self
 * next_link
 * get_submissions
 * commit_submission
 * update_submission
 * update_shippinglabel
 * driverMetadata
driverMetadata differs in naming, camel case versus snake case in other rels.

EliotSeattle, 2018-12-12:
Thanks, @winter3angle 

@balapv, have we conflated driverMetadata instances? Do the docs need to reflect something other than what they currently have for driverMetadata?

hatasum, 2018-12-21:
I have two question:
First question, my company are over 10000 products.
I use Get all products API, only can get 8910 products.
At last times, the service shows the error : "500 - Internal Server Error".

Second question, not all submission's downloads contains "drivermetadata".
By the way, how can I get all hardwareIDs and boundle ID list?
Thank you.

tedhudek, 2020-11-10:
Thanks @winter3angle for the report, and apologies for taking so long to reply.

You are right!  I have moved `driverMetadata` from the https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/get-product-data#link-object section to the https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/get-product-data#download-object section.  Also added **ExternalNotes** and **Unknown** values to the download values.

The doc update will be live in 10 minutes or so.

closed 2020-11-10 by tedhudek