MicrosoftDocs/windows-driver-docs #1143: Description of root port and endpoint power relation

tzimmer1, 2018-12-14:
Hello.
Can a blurb be added to describe the device power relationship between PCIe endpoint and root port?
We see differences in root port power state depending on if the endpoint can achieve D3Cold vs. D3Hot.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 7646563a-fd17-20f2-af5f-61ef2ce28779
* Version Independent ID: 493f9dd2-c280-a13f-8430-5676bc12b19e
* Content: [PCI driver programming guide - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/pci/index)
* Content Source: [windows-driver-docs-pr/pci/index.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/pci/index.md)
* Product: **windows-hardware**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

tedhudek, 2019-01-15:
Hi @tzimmer1 have you tried the following article?

https://docs.microsoft.com/windows-hardware/drivers/kernel/enabling-transitions-to-d3cold

tzimmer1, 2019-01-15:
Thanks Ted.  That is close but I am looking for documentation on how D3Hot and D3Cold behavior applies specifically to PCI Express root port and endpoint topologies.  A colleague of mine mentioned there may be someone at Microsoft already working on some documentation like this (Aacer?).

tedhudek, 2019-02-14:
Sorry @tzimmer1 I was unable to find anything else for you on this one.

closed 2019-02-14 by tedhudek

tzimmer1, 2019-02-21:
Not a problem.  Thanks for checking.

tzimmer1, 2019-07-16:
Try Aacer Daken (Microsoft) as a point of contact for this information.