MicrosoftDocs/windows-driver-docs #1151: Missing entry for parameter 1 = 0000000000009696

cf5146, 2018-12-18:
Microsoft (R) Windows Debugger Version 10.0.18272.1001 AMD64
Copyright (c) Microsoft Corporation. All rights reserved.

Loading Dump File [C:\WINDOWS\MEMORY.DMP]
Kernel Bitmap Dump File: Kernel address space is available, User address space may not be available.

************* Path validation summary **************
Response                         Time (ms)     Location
Deferred                                       srv*d:\symbols*https://msdl.microsoft.com/download/symbols
Symbol search path is: srv*d:\symbols*https://msdl.microsoft.com/download/symbols
Executable search path is: 
Windows 10 Kernel Version 17763 MP (8 procs) Free x64
Product: WinNt, suite: TerminalServer SingleUserTS
Built by: 17763.1.amd64fre.rs5_release.180914-1434
Machine Name:
Kernel base = 0xfffff802`53800000 PsLoadedModuleList = 0xfffff802`53c1f990
Debug session time: Tue Dec 18 10:39:02.334 2018 (UTC + 8:00)
System Uptime: 3 days 23:43:07.653
Loading Kernel Symbols
...............................................................
................................................................
................................................................
..........................
Loading User Symbols
PEB is paged out (Peb.Ldr = 00000014`68562018).  Type ".hh dbgerr001" for details
Loading unloaded module list
..................................................
For analysis of this file, run !analyze -v
nt!KeBugCheckEx:
fffff802`539b1040 48894c2408      mov     qword ptr [rsp+8],rcx ss:0018:fffff306`c582e1f0=000000000000001a
1: kd&gt; !analyze -v
*******************************************************************************
*                                                                             *
*                        Bugcheck Analysis                                    *
*                                                                             *
*******************************************************************************

MEMORY_MANAGEMENT (1a)
    # Any other values for parameter 1 must be individually examined.
Arguments:
Arg1: 0000000000009696, The subtype of the bugcheck.
Arg2: ffffbc000544fbf0
Arg3: 0000000000000000
Arg4: 0000000000000000

Debugging Details:
------------------

KEY_VALUES_STRING: 1

STACKHASH_ANALYSIS: 1

TIMELINE_ANALYSIS: 1

DUMP_CLASS: 1

DUMP_QUALIFIER: 401

BUILD_VERSION_STRING:  17763.1.amd64fre.rs5_release.180914-1434

SYSTEM_MANUFACTURER:  Dell Inc.

SYSTEM_PRODUCT_NAME:  Inspiron 7570

SYSTEM_SKU:  07EA

BIOS_VENDOR:  Dell Inc.

BIOS_VERSION:  1.11.0

[... 209 lines cut ...]
Mapping PFN database...
Mapping non paged pool...
Mapping VAD regions...
Mapping module regions...
Mapping process, thread, and stack regions...
Mapping system cache regions...

Usage:                  
Base Address:           ffffbc00`00000000
End Address:            ffffbc00`07400000
Region Size:            00000000`07400000
VA Type:                PFNDatabase

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: de955911-dfbf-33b3-778a-767a5a578a0d
* Version Independent ID: 23fb5cbd-5f7d-f7ca-1622-5885020966c2
* Content: [Bug Check 0x1A MEMORY_MANAGEMENT - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x1a--memory-management)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x1a--memory-management.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x1a--memory-management.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-01-18:
@cf5146 - Thanks for this issue. Because we have not documented information on this parameter, I would need to check with the dev team to be able to do that. We use a process to prioritize the most frequent bug checks out of the hundreds that are currently defined. With our limited documentation team resources, it is likely that we will only be able to do this work for a small sub set of all of the possible stop codes.
Right now I'm working on the new stop codes that we have not documented at all.
Thank you for your interest in improving our stop code docs.

DOMARS, 2019-03-18:
@cf5146 - I'm checking with the dev team to see if we can document the 9696 parameter 1.

DOMARS, 2019-04-15:
@cf5146 - I worked with the dev team to document 9696. Thanks for your input on the bug check topics.

closed 2019-04-15 by DOMARS