MicrosoftDocs/windows-driver-docs #1153: no revision number in device identifier after upgrading OS from win7 to win10

yingang, 2018-12-18:
According to what https://docs.microsoft.com/en-us/windows-hardware/drivers/install/identifiers-for-pci-devices says, When the Plug and Play (PnP) manager queries the driver for the hardware IDs of a device, the PCI bus driver returns a list of hardware IDs in order of increasing generality. like:

`PCI\\VEN_v(4)&DEV_d(4)&SUBSYS_s(4)n(4)&REV_r(2)`
`PCI\\VEN_v(4)&DEV_d(4)&SUBSYS_s(4)n(4)`
`PCI\\VEN_v(4)&DEV_d(4)&REV_r(2)`
`PCI\\VEN_v(4)&DEV_d(4)`
`PCI\\VEN_v(4)&DEV_d(4)&CC_c(2)s(2)p(2)`
`PCI\\VEN_v(4)&DEV_d(4)&CC_c(2)s(2)`

My question is about the middle two hardware IDs. for one PCI board developed by our company and on Win7, there would be all these six IDs. But when we upgrades the OS to Win10 (same device driver is used), only the first two and last two in the list would be displayed in device manager. Don't know where comes these differences.

Any suggestion would be apprecaited.

tedhudek, 2018-12-19:
Hi @yingang thanks for the feedback. On Win10, did the middle two IDs move to the CompatibleIds section by any chance?

tedhudek, 2018-12-19:
I actually see the same thing with the display adapter on my Win10 machine.  Four values appear in Hardware Ids, and 2 values are in compatible Ids.  I've added a note to the doc page.  I'm going to close the issue, but please feel free to re-open if you have more questions or problems.  Thanks again!

closed 2018-12-19 by tedhudek