MicrosoftDocs/windows-driver-docs #1160: What about debugging an ELAM service?

BrianCatlin, 2018-12-20:
This page (https://docs.microsoft.com/en-us/windows/desktop/Services/protecting-anti-malware-services-) doesn't provide enough info and should really be in this section because you cannot have an ELAM service without an ELAM driver. Also, I have a C# ELAM service, and I cannot get the debugger to attach to it. What do I need to do?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 6cb04a99-d20e-4b9b-9c25-bbeaaed52b8d
* Version Independent ID: 80858d16-aa19-4992-1767-4c1bc365f297
* Content: [Early Launch AntiMalware - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/early-launch-antimalware)
* Content Source: [windows-driver-docs-pr/install/early-launch-antimalware.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/early-launch-antimalware.md)
* Product: **windows-hardware**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2019-01-09:
Hi @BrianCatlin because the service runs as PPL (Protected Process Light, described at https://docs.microsoft.com/en-us/windows/desktop/services/protecting-anti-malware-services-#debugging-an-anti-malware-protected-service) you need to debug via kernel debugger.  Does that help?

BrianCatlin, 2019-01-09:
I tried that, but I couldn’t get a breakpoint- even putting in a hard break, so I had to rewrite everything in C++

closed 2019-01-14 by tedhudek