MicrosoftDocs/windows-driver-docs #1176: How can we achieve it with a single computer?

cshivaprasad, 2018-12-24:
Can u explain, how to test with a single computer

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 287160ad-f2a5-789a-2d7d-ff336514c11e
* Version Independent ID: a138162d-2b80-725b-212b-277bae8f177d
* Content: [Write a Universal Windows driver (UMDF 2) based on a template - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/gettingstarted/writing-a-umdf-driver-based-on-a-template)
* Content Source: [windows-driver-docs-pr/gettingstarted/writing-a-umdf-driver-based-on-a-template.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/gettingstarted/writing-a-umdf-driver-based-on-a-template.md)
* Product: **windows-hardware**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

EliotSeattle, 2019-01-11:
Hi @cshivaprasad 
You really should be using a second PC. When you debug the driver using the kernel debugger, you can lock/dead-stop the testing PC, as you are looking at memory contents and single stepping through the kernel. That will render the single computer unusable. 
It is also common to encounter a blue screen on the testing PC as the code is being developed. That also makes the single computer unusable.
Hope this helps.

closed 2019-01-11 by EliotSeattle

ericleigh007, 2019-01-25:
If you provision manually, you can use a VM for these types of tests.