MicrosoftDocs/windows-driver-docs #1182: Not working -logo options

kozera2137, 2018-12-27:
Hi. I wanted to specify WinDbg logging file, however it seems to doesn't work.
`PS C:\Windows\system32> WinDbgX.exe -logo C:\log.txt`
I was running with administrator privileges and trying other path but without results. Process Monitor doesn't even see CreateFile call.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 8353dbe7-846a-f902-0f6c-f7e75ad846d8
* Version Independent ID: 6ec7d01f-69f9-5f7b-582b-0d63d9c40a06
* Content: [WinDbg Preview - Command line startup options - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/windbg-command-line-preview)
* Content Source: [windows-driver-docs-pr/debugger/windbg-command-line-preview.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/windbg-command-line-preview.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2018-12-28:
@aluhrs13 - I tried this as well and I'm not seeing a log created when the debugger is launched like this:
```
windbgx /logo c:\log.txt
```

DOMARS, 2019-01-26:
@kozera2137 - The dev team is looking at this issue. Thanks for passing on what you encountered.

DOMARS, 2019-10-07:
@kozera2137 - My understanding is that this should work, so I believe this to be a temporary bug with the product, not a docs issue, so closing this doc issue. To track issues with Windbg Preview, use this feedback channel:

http://aka.ms/dexex 

@aluhrs13 - FYI

Thanks for your interest in the debugger and the docs.

closed 2019-10-07 by DOMARS