MicrosoftDocs/windows-driver-docs #1187: bcdedit /Set misses the warning for Powershell

Karl-WE, 2018-12-31:
when you lookup bcdedit set there is a warning for users that use PowerShell.

ID e.g. {current} has to be set in " " using PowerShell. This note is missing here in the examples. Please include the advice  according to 

https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/bcdedit--set

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: b4e56d98-d380-e346-4ba3-d17f1596af86
* Version Independent ID: 2946df6a-59c0-335a-9a6a-5e9ceabaf13f
* Content: [Changing the Friendly Name of a Boot Entry - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/changing-the-friendly-name-of-a-boot-entry)
* Content Source: [windows-driver-docs-pr/devtest/changing-the-friendly-name-of-a-boot-entry.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/changing-the-friendly-name-of-a-boot-entry.md)
* Product: **windows-hardware**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

DOMARS, 2019-01-02:
@Karl-WE - Thanks for the suggestion, I added the same text to the topic and moved BCDEdit up to the top of the topic as that is more popular these days. The updates should be live in a day or so. 

You are always welcome to submit a pull request.

DOMARS, 2019-01-03:
@Karl-WE - The changes are live, so closing this issue. Thanks for the input.

closed 2019-01-03 by DOMARS