MicrosoftDocs/windows-driver-docs #1197: !pool not working anymore with Windows 10 1809

shaharlux, 2019-01-07:
!pool command is one of the most important commands. The fact that it is no longer working for Windows 10 1809 is a real problem when debugging Windows 10 Kernel issues. It needs to be fixed as quickly as possible.
Currently, output of !pool command is: "Failed to read heap key"

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 23dacdcf-6f0a-4203-f449-f335ebafef1d
* Version Independent ID: d7a4e58b-4453-b8f6-826a-dbf97e0bd453
* Content: [pool extension command - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/-pool)
* Content Source: [windows-driver-docs-pr/debugger/-pool.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/-pool.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-01-07:
@shaharlux - I'm seeing issues as well, I wonder if something has changed here @aluhrs13?

I'm testing in WinDbg Preview version 1.0.1812.12001 with engine 10.0.18303. 

Target Windows OS is 1809. 

``` 
0: kd> vertarget
Windows 10 Kernel Version 17763 MP (4 procs) Free x64
Product: WinNt, suite: TerminalServer SingleUserTS
Built by: 17763.1.amd64fre.rs5_release.180914-1434
```

```
0: kd> !pool -1
Pool page ffffffffffffffff region is Unknown
fffffffffffff000 is not a valid large pool allocation, checking large session pool...
Unable to read large session pool table (Session data is not present in mini and kernel-only dumps)
fffffffffffff000 is not valid pool. Checking for freed (or corrupt) pool
Address fffffffffffff000 could not be read. It may be a freed, invalid or paged out page

0: kd> !poolfind -tag "ACPI"

Scanning large pool allocation table for tag 0x49504341 (ACPI) (ffff800483c10000 : ffff800483d90000)

ERROR: Failed to read from page segment ffff8003eaa00000
ERROR: Failed to read from page segment ffffd78e02d00000
ERROR: Failed to read from page segment ffff80047e700000
```
!poolused seems to work.

```
0: kd> !poolused
....
 Sorting by Tag

               NonPaged                  Paged
 Tag     Allocs         Used     Allocs         Used

 ACPI         4          640          0            0       UNKNOWN pooltag 'ACPI', please update pooltag.txt
 ALPC      3142      1871136          0            0       ALPC port objects , Binary: nt!alpc
 APpt      2007       257104       4389       445680       UNKNOWN pooltag 'APpt', please update pooltag.txt
 ARFT         0            0          3          192       UNKNOWN pooltag 'ARFT', please update pooltag.txt
 AVNw         1         1408          0            0       UNKNOWN pooltag 'AVNw', please update pooltag.txt
 AVns         1           48          0            0       UNKNOWN pooltag 'AVns', please update pooltag.txt
 AVpc         1         1408          0            0       UNKNOWN pooltag 'AVpc', please update pooltag.txt
 AVpd         1         1408          0            0       UNKNOWN pooltag 'AVpd', please update pooltag.txt
 AVwi         1          672          0            0       UNKNOWN pooltag 'AVwi', please update pooltag.txt
 AcpA         2          160          0            0       ACPI arbiter data , Binary: acpi.sys
 AcpB         0            0          1           48       ACPI buffer data , Binary: acpi.sys
 AcpD        88        81872          0            0       ACPI device data , Binary: acpi.sys
```

DOMARS, 2019-01-07:
@shaharlux - You may also want to follow this link to file product feedback using the Windows Feedback Hub: 
http://aka.ms/dexex 
The Feedback Hub allows customers to file and up vote product issues. The link automatically sets the product category to Debugging Tools for Windows.

shaharlux, 2019-01-08:
Thanks you @DOMARS. I've send feedback with the Feedback Hub.

474172261, 2019-05-22:
Has it been fixed?? I'm tired of this bug. :(

rodwiddowson, 2019-05-22:
+1 to that.

Also the link that @DOMARS provided above isn't working for me and an attempt to go to the root just throws me into a twisty little maze Azure-AD passages, all different.

DOMARS, 2019-05-22:
@rodwiddowson - You likely need to be logged in.

Here is another place that you can provide product input, if the feedback tool is not working for you:

https://github.com/microsoftfeedback/WinDbg-Feedback

DOMARS, 2019-06-05:
@rodwiddowson - As this product issue is being tracked in https://github.com/microsoftfeedback/WinDbg-Feedback/issues/3
I'm closing the docs issue. Thanks for your interest in the debugger and working with the product team on this issue with !pool.

closed 2019-06-05 by DOMARS