MicrosoftDocs/windows-driver-docs #1201: PIPE_TRANSFER_TIMEOUT timeout in what units? [nano/micro/milli]seconds?

leo-zspace, 2019-01-08:
timeout in what units in call:
WinUsb_SetPipePolicy(winusb_handle, endpoint_id, PIPE_TRANSFER_TIMEOUT, sizeof(ULONG), &timeout))

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 947f85b5-07bc-45eb-4bb9-75302b0566b6
* Version Independent ID: 73454115-5c1c-a8ea-f65d-0a40116cb58b
* Content: [WinUSB Functions for Pipe Policy Modification - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/winusb-functions-for-pipe-policy-modification)
* Content Source: [windows-driver-docs-pr/usbcon/winusb-functions-for-pipe-policy-modification.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/usbcon/winusb-functions-for-pipe-policy-modification.md)
* Product: **windows-hardware**
* GitHub Login: @duncanmacmichael
* Microsoft Alias: **dumacmic**

duncanmacmichael, 2019-01-08:
Hi @leo-zspace, good question. Let me ask the feature team so I can update the table in the doc accordingly.

duncanmacmichael, 2019-01-14:
Still working on getting an answer for you, thanks for your patience.

leo-zspace, 2019-01-14:
I actually know the answer (but of course - it is "easy" to get experimentally) and confirmed it online here:
https://android.googlesource.com/platform/development/+/master/host/windows/usb/winusb/adb_winusb_endpoint_object.h#138
and here:
https://android.googlesource.com/platform/development/+/master/host/windows/usb/winusb/adb_winusb_endpoint_object.cpp
(obviously thru experimentation too).

But it is good to have it documented (this is what documentation is for, isn't it?) and carve in stone and ensure to some degree that it's not going to change to something different w/o any notice.

Thank you for taking time and putting effort into it.

duncanmacmichael, 2019-01-14:
Great, glad you found an answer. Little details like this can really make the difference on the docs. :) I'll clarify that the timeout value is indeed in milliseconds, and that change should be live later today if not tomorrow. Thanks!

closed 2019-01-14 by duncanmacmichael