MicrosoftDocs/windows-driver-docs #1202: Causes for this error

LuigiBruno, 2019-01-08:
Add a list of the possible causes for this error.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 3bcd11f1-65ff-4b7c-39ad-bcdade7050a2
* Version Independent ID: 4c33cb17-d2ad-f119-a16a-aa47c6e5d2f2
* Content: [Bug Check 0x2 DEVICE_QUEUE_NOT_BUSY - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x2--device-queue-not-busy)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x2--device-queue-not-busy.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x2--device-queue-not-busy.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-01-17:
@LuigiBruno - Thanks for this issue. As I have mentioned before, we use a process to prioritize the most frequent bug checks out of the hundreds that are currently defined. With our limited documentation team resources, it is likely that we will only be able to do this work for a small sub set of all of the possible stop codes.
Right now I'm working on the new stop codes that we have not documented at all.
Thank you for your interest in improving our stop code docs.

DOMARS, 2019-09-11:
@LuigiBruno - I have opened an internal work item to track this work, so closing this issue for now.

As I have mentioned before, we use a process to prioritize the most frequent bug checks out of the hundreds that are currently defined. With our limited documentation team resources, it is likely that we will only be able to do this work for a small sub set of all of the possible stop codes.

Thank you for your interest in improving our stop code docs.

closed 2019-09-11 by DOMARS