MicrosoftDocs/windows-driver-docs #1208: Add causes

LuigiBruno, 2019-01-08:
Add the "Causes" section to indicate possible reasons for this bug check to occur.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 815c73da-d89f-62fa-5ba5-67e079395e03
* Version Independent ID: 5daba5f7-0540-6225-4f91-9b663c5f8372
* Content: [Bug Check 0x4 INVALID_DATA_ACCESS_TRAP - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x4--invalid-data-access-trap)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x4--invalid-data-access-trap.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x4--invalid-data-access-trap.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-01-17:
@LuigiBruno - Thanks for this issue.  As I have mentioned before, we use a process to prioritize the most frequent bug checks out of the hundreds that are currently defined. With our limited documentation team resources, it is likely that we will only be able to do this work for a small sub set of all of the possible stop codes. 

Right now I'm working on the new stop codes that we have not documented at all.

Thank you for your interest in improving our stop code docs.

DOMARS, 2019-09-11:
@LuigiBruno - I have opened an internal work item to track this work, so closing this issue for now.

As I have mentioned before, we use a process to prioritize the most frequent bug checks out of the hundreds that are currently defined. With our limited documentation team resources, it is likely that we will only be able to do this work for a small sub set of all of the possible stop codes.

Thank you for your interest in improving our stop code docs.

closed 2019-09-11 by DOMARS

LuigiBruno, 2019-10-09:
> @LuigiBruno - I have opened an internal work item to track this work, so closing this issue for now.
> As I have mentioned before, we use a process to prioritize the most frequent bug checks out of [...]

Thank you @DOMARS.